technik-gegg/SMuFF-1.1 #1: Use SMuFF with Klipper Firmware

troisplus, 2019-07-11:
Hi, 
I wanna know if you already thought about klipper compatibility. It combines the power of a raspberry Pi 3 with one or more micro-controllers.
(https://github.com/KevinOConnor/klipper)

Actually I use a Anet A8 running klipper with an arduino mega/ramps shield. I used marlin few month ago but had poor quality prints.
I hope to install a MMU system with multi controlers on RPi3

technik-gegg, 2019-10-13:
Hi troisplus,
sorry for the late reply. I wasn't ignoring your request, I simply got no notification about it.

No, I havent thought  about Klipper as of yet but that's not really necessary.
I wrote the code for the SMuFF in a way that will make it easy for any firmware to support the SMuFF as long as it can handle a serial port and can talk GCode (which eventually means: all of them ;o).

So, if a firmware wants to utilize the SMuFF, it simply sends the Tx command over the serial interface and the SMuFF will do the rest.

By looking at the code you sent me in the link, I can't see any support for tool changing. But as much (or little) as I know about Klipper, I think it's configurable somewhere. 
Do you have any links of samples for other tool changing hardware supported by Klipper?
If so, please let me know and I'll take a look.

closed 2019-12-14 by technik-gegg

flopana77, 2020-10-15:
Hi technik-geeg
I am also looking at running the SMuFF via Klipper. That said, I wanted to build the V5 and didn't put my mail address in the donation...Is there any way we can get in contact regarding the stl-files and to discuss the Klipper implementation later on?
Cheers

technik-gegg, 2020-10-15:
I've already sent you an email with the files.

As far as it concerns Klipper, I haven't had the chance yet to set it all up (with the latest Klipper modifications). Though, there's at least one member on the Discord channel (Carpediem1903) who's after this as well. 
You're invited to join and discuss along.