technik-gegg/SMuFF-1.1 #2: error: 'uint32' does not name a type

paulpenney, 2019-11-29:
Hello!

Just getting this error when I am trying to compile for the Wanhao board. Tried on both VScode and Atom, but the same issue. Any ideas?

paulpenney, 2019-11-29:
Here is the terminal: 

Processing WANHAO_I3_MINI (platform: atmelavr; framework: arduino; board: megaatmega2560)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/megaatmega2560.html
PLATFORM: Atmel AVR 2.0.0 > Arduino Mega or Mega 2560 ATmega2560 (Mega 2560)
HARDWARE: ATMEGA2560 16MHz, 8KB RAM, 248KB Flash
PACKAGES: framework-arduino-avr 5.0.0, toolchain-atmelavr 1.50400.190710 (5.4.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for U8G2 library in registry
Found: https://platformio.org/lib/show/942/U8g2
LibraryManager: Installing id=942
U8g2 @ 2.27.2 is already installed
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SdFs> 2018.6.24
|   |-- <SPI> 1.0
|-- <ArduinoJson> 6.13.0
|-- <MemoryFree> #a2d60bd
|-- <Wire> 1.0
|-- <SPI> 1.0
|-- <U8g2> 2.27.2
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
Building in release mode
Compiling .pio\build\WANHAO_I3_MINI\src\ClickEncoder.cpp.o
In file included from src\ClickEncoder.cpp:12:0:
include/ClickEncoder.h: In constructor 'ClickEncoder::ClickEncoder(uint8_t, uint8_t, uint8_t, uint8_t, bool)':
include/ClickEncoder.h:116:29: warning: 'ClickEncoder::doubleClickEnabled' will be initialized after [-Wreorder]
   bool doubleClickEnabled = false;  
                             ^
include/ClickEncoder.h:109:31: warning:   'bool ClickEncoder::accelerationEnabled' [-Wreorder]
   bool  accelerationEnabled = false;
                               ^
src\ClickEncoder.cpp:46:1: warning:   when initialized here [-Wreorder]
 ClickEncoder::ClickEncoder(uint8_t A, uint8_t B, uint8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
In file included from src\ClickEncoder.cpp:12:0:
include/ClickEncoder.h:109:31: warning: 'ClickEncoder::accelerationEnabled' will be initialized after [-Wreorder]
   bool  accelerationEnabled = false;
                               ^
include/ClickEncoder.h:106:28: warning:   'volatile int16_t ClickEncoder::delta' [-Wreorder]
   volatile int16_t delta = 0;
                            ^
src\ClickEncoder.cpp:46:1: warning:   when initialized here [-Wreorder]
 ClickEncoder::ClickEncoder(uint8_t A, uint8_t B, uint8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
In file included from src\ClickEncoder.cpp:12:0:
include/ClickEncoder.h:113:28: warning: 'ClickEncoder::button' will be initialized after [-Wreorder]
   volatile Button button = Open;
                            ^
include/ClickEncoder.h:110:19: warning:   'uint8_t ClickEncoder::steps' [-Wreorder]
   uint8_t steps = 0;
                   ^
src\ClickEncoder.cpp:46:1: warning:   when initialized here [-Wreorder]
 ClickEncoder::ClickEncoder(uint8_t A, uint8_t B, uint8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
In file included from src\ClickEncoder.cpp:12:0:
include/ClickEncoder.h:110:19: warning: 'ClickEncoder::steps' will be initialized after [-Wreorder]
   uint8_t steps = 0;
                   ^
include/ClickEncoder.h:102:17: warning:   'const uint8_t ClickEncoder::pinA' [-Wreorder]
   const uint8_t pinA;
                 ^
src\ClickEncoder.cpp:46:1: warning:   when initialized here [-Wreorder]
 ClickEncoder::ClickEncoder(uint8_t A, uint8_t B, uint8_t BTN, uint8_t stepsPerNotch, bool active)
 ^
[... 65 lines cut ...]
src\Menus.cpp: In function 'void setupMainMenu(char*)':
src\Menus.cpp:70:10: warning: unused variable 'items2' [-Wunused-variable]
     char items2[150];
          ^
src\Menus.cpp: In function 'void showOffsetsMenu(char*)':
src\Menus.cpp:989:9: warning: variable 'title' set but not used [-Wunused-but-set-variable]
   char* title;
         ^
src\SMuFFtools.cpp: In function 'bool setServoPos(int, int)':
src\SMuFFtools.cpp:1226:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
*** [.pio\build\WANHAO_I3_MINI\src\SMuFF.cpp.o] Error 1
*** [.pio\build\WANHAO_I3_MINI\src\GCodes.cpp.o] Error 1
*** [.pio\build\WANHAO_I3_MINI\src\SMuFFtools.cpp.o] Error 1
=========================================================================================================== [FAILED] Took 6.03 seconds ===========================================================================================================

Environment       Status    Duration
----------------  --------  ------------
WANHAO_I3_MINI    FAILED    00:00:06.033
BIGTREE_SKR_MINI  IGNORED
===================================================================================================== 1 failed, 0 succeeded in 00:00:06.033 =====================================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

technik-gegg, 2019-12-10:
Yeah, sorry for that and for not seeing your post.

I've updated the source code.

closed 2019-12-14 by technik-gegg