technik-gegg/SMuFF-1.1 #4: the display does not work

NamkingWIN, 2019-12-16:
Hi,
I have tried complied the firmware and flash it to SKR mini using stlinkV2 module. After that, I connect the oled lcd 0.96inch to skr (EXP1), the oled does not work, still dark. How can I check the SKR is still work properly?
Here is pic of my problem: https://app.box.com/s/7yo3w6t78riltsedot4lnrkzcf71cbjx
Thanks

technik-gegg, 2019-12-16:
There are a couple of methods you can try:
1. Connect a buzzer to it (EXP1/PC10). It'll play a tune on startup.
2. Connect a stepper motor to the Revolver. It's supposed to go to home position after startup. If you don't have an endstop on it, it'll turn infinitely. Trigger the endstop and it'll stop.
3. Start debugging the firmware from within PlatformIO and set a break point in setup(). It'll use the ST-Link to control/communicate.

Usually, if the programming was successful (SUCCESS response in PlatformIO) you have to check the wiring of your LCD. Since there are only 4 wires, of which 2 of them are powering the display you only have to check the SDA/SCL wires. Power wiring seems ok.

NamkingWIN, 2019-12-16:
I have debug and the result is below
`Processing BIGTREE_SKR_MINI (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for U8G2 library in registry
Found: https://platformio.org/lib/show/942/U8g2
LibraryManager: Installing id=942
U8g2 @ 2.27.3 is already installed
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SdFs> 2018.6.24
|   |-- <SPI> 1.0
|-- <ArduinoJson> 6.13.0
|-- <MemoryFree> #a2d60bd
|-- <U8g2> 2.27.3
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in debug mode
Checking size .pio\build\BIGTREE_SKR_MINI\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [==        ]  22.1% (used 10856 bytes from 49152 bytes)
PROGRAM: [======    ]  63.5% (used 166460 bytes from 262144 bytes)
========================= [SUCCESS] Took 6.58 seconds =========================

Environment       Status    Duration
----------------  --------  ------------
WANHAO_I3_MINI    IGNORED
BIGTREE_SKR_MINI  SUCCESS   00:00:06.579
========================= 1 succeeded in 00:00:06.579 =========================
Reading symbols from d:\STL files\SMuFF+-+Smart+Multi+Filament+Feeder+with+#Bondtech+gears\SKR firmware\SMuFF-1.1\.pio\build\BIGTREE_SKR_MINI\firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = stlink
PlatformIO: Initializing remote target...
xPack OpenOCD, 32-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-07:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 1000 kHz
Info : STLINK V2J34S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.252695
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
target halted due to debug-request, current mode: Handler External Interrupt(32)
xPSR: 0x01000030 pc: 0x08001654 msp: 0x2000be34
Info : device id = 0x10036414
Info : flash size = 256kbytes
0x08001654 in dma_clear_isr_bits (tube=DMA_CH5, dev=0x40004400) at C:\users\nguyen hoai nam\.platformio\packages\framework-arduinoststm32-maple\STM32F1\system\libmaple/stm32f1/include/series/dma.h:538
538	    dev->regs->IFCR = (1U << (4 * (tube - 1)));
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080003f0 msp: 0x2000c000
Loading section .text, size 0x21bcc lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x8021bd0
Loading section .data, size 0x1218 lma 0x8021bd8
Loading section .rodata, size 0x5c4c lma 0x8022df0
Info : Padding image section 0 at 0x08021bcc with 4 bytes
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000454 msp: 0x2000c000
Start address 0x8000454, load size 166456
Transfer rate: 22 KB/sec, 12804 bytes/write.
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000454 msp: 0x2000c000
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000454 msp: 0x2000c000
Temporary breakpoint 1 at 0x8004882: file C:\users\nguyen hoai nam\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\main.cpp, line 38.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.

Temporary breakpoint 1, main () at C:\users\nguyen hoai nam\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\main.cpp:38
38	    setup();
`

NamkingWIN, 2019-12-16:
Here is my process of program the SKR:
1. Compile the firmware using platformIO
2. Upload firmware using platformIO after compiled success.
 Are wrongs in my process?

technik-gegg, 2019-12-16:
Nothing wrong with this. As far as I can tell, firmware is working. Debugger has stopped in main(), which it usually does. If you step through, it'll call setup() next. Try to step through to **setupSteppers()** in setup(). If that's working, check wiring of the display or try a different one.

NamkingWIN, 2019-12-16:
Here is the message after compiled firmware 
`Linking .pio\build\BIGTREE_SKR_MINI\firmware.elf
Building .pio\build\BIGTREE_SKR_MINI\firmware.bin
Checking size .pio\build\BIGTREE_SKR_MINI\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [==        ]  22.1% (used 10856 bytes from 49152 bytes)
PROGRAM: [======    ]  63.5% (used 166460 bytes from 262144 bytes)
============================================= [SUCCESS] Took 109.93 seconds =============================================

Environment       Status    Duration
----------------  --------  ------------
WANHAO_I3_MINI    IGNORED
BIGTREE_SKR_MINI  SUCCESS   00:01:49.926
============================================== 1 succeeded in 00:01:49.926 ==============================================

Terminal will be reused by tasks, press any key to close it.`

NamkingWIN, 2019-12-16:
And this is the message Upload firmware 

`Executing task: C:\Users\Nguyen Hoai Nam\.platformio\penv\Scripts\platformio.exe run --target upload --environment BIGTREE_SKR_MINI <

Processing BIGTREE_SKR_MINI (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
-------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2, tool-openocd 2.1000.190707 (10.0), tool-dfuutil 1.9.190708
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for U8G2 library in registry
Found: https://platformio.org/lib/show/942/U8g2
LibraryManager: Installing id=942
U8g2 @ 2.27.3 is already installed
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SdFs> 2018.6.24
|   |-- <SPI> 1.0   
|-- <ArduinoJson> 6.13.0
|-- <MemoryFree> #a2d60bd
|-- <U8g2> 2.27.3
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in release mode
Checking size .pio\build\BIGTREE_SKR_MINI\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [==        ]  22.1% (used 10840 bytes from 49152 bytes)
PROGRAM: [=====     ]  50.8% (used 133108 bytes from 262144 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\BIGTREE_SKR_MINI\firmware.elf
xPack OpenOCD, 32-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-07:34)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08000454 msp: 0x2000c000
** Programming Started **
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
============================================= [SUCCESS] Took 15.45 seconds =============================================

Environment       Status    Duration
----------------  --------  ------------
WANHAO_I3_MINI    IGNORED
BIGTREE_SKR_MINI  SUCCESS   00:00:15.448
============================================== 1 succeeded in 00:00:15.448 ==============================================

Terminal will be reused by tasks, press any key to close it.`

NamkingWIN, 2019-12-16:
> Nothing wrong with this. As far as I can tell, firmware is working. Debugger has stopped in main(), which it usually does. If you step through, it'll call setup() next. Try to step through to **setupSteppers()** in setup(). If that's working, check wiring of the display or try a different one.

Here is the message after step through setup()

`Processing BIGTREE_SKR_MINI (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for U8G2 library in registry
Found: https://platformio.org/lib/show/942/U8g2
LibraryManager: Installing id=942
U8g2 @ 2.27.3 is already installed
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SdFs> 2018.6.24
|   |-- <SPI> 1.0
|-- <ArduinoJson> 6.13.0
|-- <MemoryFree> #a2d60bd
|-- <U8g2> 2.27.3
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Wire> 1.0
Building in debug mode
Checking size .pio\build\BIGTREE_SKR_MINI\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [==        ]  22.1% (used 10856 bytes from 49152 bytes)
PROGRAM: [======    ]  63.5% (used 166460 bytes from 262144 bytes)
========================= [SUCCESS] Took 8.12 seconds =========================

Environment       Status    Duration
----------------  --------  ------------
WANHAO_I3_MINI    IGNORED
BIGTREE_SKR_MINI  SUCCESS   00:00:08.117
========================= 1 succeeded in 00:00:08.117 =========================
Reading symbols from d:\STL files\SMuFF+-+Smart+Multi+Filament+Feeder+with+#Bondtech+gears\SKR firmware\SMuFF-1.1\.pio\build\BIGTREE_SKR_MINI\firmware.elf...
done.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = stlink
PlatformIO: Initializing remote target...
xPack OpenOCD, 32-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-07:34)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : tcl server disabled
Info : telnet server disabled
Info : clock speed 1000 kHz
Info : STLINK V2J34S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.252695
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : accepting 'gdb' connection from pipe
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08001852 msp: 0x2000bf50
Info : device id = 0x10036414
Info : flash size = 256kbytes
0x08001852 in gpio_set_mode (dev=<optimized out>, pin=0 '\000', mode=1) at C:\users\nguyen hoai nam\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\libmaple\gpio_f1.c:128
128	    uint32 tmp = *cr;
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x080003f0 msp: 0x2000c000
Loading section .text, size 0x21bcc lma 0x8000000
Loading section .ARM.exidx, size 0x8 lma 0x8021bd0
Loading section .data, size 0x1218 lma 0x8021bd8
Loading section .rodata, size 0x5c4c lma 0x8022df0
Info : Padding image section 0 at 0x08021bcc with 4 bytes
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000454 msp: 0x2000c000
Start address 0x8000454, load size 166456
Transfer rate: 22 KB/sec, 12804 bytes/write.
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000454 msp: 0x2000c000
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000454 msp: 0x2000c000
Temporary breakpoint 1 at 0x8004882: file C:\users\nguyen hoai nam\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\main.cpp, line 38.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Note: automatically using hardware breakpoints for read-only addresses.

Temporary breakpoint 1, main () at C:\users\nguyen hoai nam\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\main.cpp:38
38	    setup();

Breakpoint 2, setup () at src\SMuFF.cpp:172
172	void setup() {
Info : halted: PC: 0x08011528
halted: PC: 0x08011528
`

NamkingWIN, 2019-12-16:
I have check the oled, it works fine with the wemos d1 mini using U8g2lib. But not work with SKR.

technik-gegg, 2019-12-16:
Check your wiring according these pictures.
[image]
[image]

NamkingWIN, 2019-12-16:
> Check your wiring according these pictures.
> [image]
> [image]

My wiring is the same with you.
https://app.box.com/s/cr6c3lm7raa5h316p43rlikrf52rujwb
https://app.box.com/s/z9n6lnp9wq1298xhezy7aitq3f5e1xou
https://app.box.com/s/0yzkiba4ywawvohywzupagmqm42uylwk

technik-gegg, 2019-12-16:
I don't see an SD-Card in your SKR. You have to copy the according configuration file (SMUFF.CFG) onto a SD-Card and stick that into the board. Otherwise it'll not  work as expected.

NamkingWIN, 2019-12-16:
> I don't see an SD-Card in your SKR. You have to copy the according configuration file (SMUFF.CFG) onto a SD-Card and stick that into the board. Otherwise it'll not work as expected.

Where can I get this file? I saw the SMUFF-DUET.CFG at the Config/SKR mini/. Is that file you mean?

technik-gegg, 2019-12-16:
Yep. Rename it to SMUFF.CFG after copying.

NamkingWIN, 2019-12-16:
> Yep. Rename it to SMUFF.CFG after copying.

OK, thanks. The Oled work well. 
The speaker has loud at begin. But the revolver step just turn 1 circle and stop not turn infinitely as you said.!!!
Another problem: the rotary encoder not affect when I scroll it. I am using the breaked out rotary encoder (intergrated 2 resistor 10k) as pic below.
https://app.box.com/s/ob6rpv6b7imgurjk8l9m53c2j3s6pqn5
My wiring is: 
CLK --> PC14
DT --> PC15
SW -->PC11
+ --> 5V
GND --> GND

technik-gegg, 2019-12-16:
Don't know. Please double check your wiring.

NamkingWIN, 2019-12-17:
I am check the wiring of encoder, just as your setup. And I also check the encoder work fine, switch button of encoder can reset the board but not affect when connected at right place. Some problem in compile firmware (pic below) may cause the error?
https://app.box.com/s/6xevlji548lz9mh2ejrs88wef4he3jln

technik-gegg, 2019-12-17:
Those are no errors, just warnings.

The code will compile and the encoder does work! If not, you did something wrong assembling it.

NamkingWIN, 2019-12-17:
I just wiring the oled, speaker, reset button, encoder and the step with driver. That is enough for testing of SKR work properly? Or must wiring more component? Maybe the revolver turn to reach to the endstop at the begin so encoder does not work temporarily, I thought that. Is that right?

NamkingWIN, 2019-12-18:
Here are my wiring, please look pic at link below
https://app.box.com/s/wpir4ybonsuonxbh78x09zpzzs0ubi81
With this wiring the oled, speaker and reset button work fine but the encoder not work.
At the begin, the step of revolver just turn only 1 round and stop. (you can see at video https://app.box.com/s/aadpcgdieynzhfc6mzdi0ohm14macqzj)
Please help me to check the error. Thanks so much.

NamkingWIN, 2019-12-19:
> Disable the Revolver by setting "UseServo" to **true** in **SMuFF.CFG** on your SD-Card (section Revolver) and try it again.
> If that doesn't change anything measure the voltage on the Encoder.
> 
> Basically it's never a good idea to half assemble something and expect it to run. Finish it up first, then try things out.
> 
> And btw: The pull-up resistors are missing in your drawing!

Enable revolver to using servo, oled indicate "reading config" and stop there. I have still collecting some items to assembly SMuFF but it will complete soon.
Can you add some message show on the oled for debuging or notice to user where error is occured?

technik-gegg, 2019-12-19:
If you get the "reading config", your SMuFF.cfg is corrupted. You probably missed a coma.
Open the file in an editor, copy & paste it in here: https://arduinojson.org/v6/assistant/
It'll tell you what's wrong with it. Correct it and try again.  
Unless the config is not read fully, there's not much to say debug wise.

technik-gegg, 2019-12-19:
Closing this thread because it doesn't relate to the issue anymore.

closed 2019-12-19 by technik-gegg