technik-gegg/SMuFF-1.1 #5: Maybe booting failed even oled show ready!

NamkingWIN, 2019-12-24:
Hi,
On the oled show “I ready” but at the left-down corner show “0” instead the free memory left. What I am getting error?

technik-gegg, 2019-12-24:
That's not an error, that's by design. The memory display reliably only works on the ATMega CPUs but not on STM32. Hence, this information has been replaced by the amount of filament feed on the SKR board.
It's all as it's supposed to be.

codycato, 2019-12-29:
I'm at the same point now. Long press doesnt work for me.

technik-gegg, 2019-12-29:
Well guys, again, check your wiring!

This is, what it's supposed to be: 

[image]

I'm pretty sure you've left out some connections.

codycato, 2019-12-29:
In the pic you have of the lcd board. The  Rotary knob section...S1= pin 9/BT-ENC. And  S2= GND.

The pic above shows the opposite. I tried both but same results.

technik-gegg, 2019-12-29:
That doesn't matter at all. It's just a switch that shorts to GND. Will work either way.

technik-gegg, 2019-12-29:
Maybe the flat ribbon cable you're using to connect to the board sucks. Try replacing it.

codycato, 2019-12-29:
Yea I thought about that so i tried a secon ribbon and and also went straight dupont to the skr connector. I'll try another rotary encoder tomorrow. 

As for the SMUFF.cfg parser check says valid but it never boots whenever I edit and save. Only works when I just rename your original file.

technik-gegg, 2019-12-29:
I've uploaded a CFG for the servo variant. Try this one.

What does the ArduinoJson assistant say about the Memory Pool Size for ESP32/ESP8266? It's supposed to be somewhere around 1764 bytes.

codycato, 2019-12-29:
I'm not at the computer right now but If I remember correctly it was +1800. Can you upload cfg servo for duet please.

technik-gegg, 2019-12-29:
Did that.

codycato, 2019-12-29:
Stuck at "reading config" with the file you just uploaded :(

technik-gegg, 2019-12-29:
Try a different SD-Card, formatted with standard FAT.

codycato, 2019-12-29:
Did that, still stuck. I only have 2 sdcard that can format with standard FAT

codycato, 2019-12-29:
This is so weird. The original SMUFF_DUET.CFG works when I rename it. But the new ones dont.

technik-gegg, 2019-12-29:
Yep, that's weird... I just did the same without any issues.

closed 2019-12-30 by technik-gegg