technik-gegg/SMuFF-1.1 #6: firmware.bin

davidcgu, 2020-01-03:
Hello mate,

First of all wanted to thank you for your great job and sharing, thats really amazing!!!!!!

I'm on the way of building mine following your instructions however I have a small problem....

I ordered an SKR Mini which unfortunately the micro SD card reader is not working, I'm claming to te seller to get a replacement however this will take no less than a month what will delay a lot my follow up....

I have made the test with an ST-link and I'm able to load the bootloader and the firmware.bin for marlin and it works as I can see as result that boots fine with marlin screen on an 2004 LCD.

I have my controler board made and ready to test with I2C address 60, I have compiled and flashed on the same way I do with the maling firmware.bin but nothing is shown on the oled display afterwards.

I assume it does not work as is not able to read the config flle on the micro sd card.

I wonder if provisionally there is some way to configure some firm that I can modify directly on VS and compile and run away from the sd card to make it boot and make some first motion tests.

Thanks in advance for your attention.

Aps... Happy new year to everybody!!!!!!!

technik-gegg, 2020-01-03:
The only thing you can do without an SD-Card is to hard code all values into the **SMuFFConfig** structure in **SMuFF.h** and comment out the readConfig() function in SMuFF.cpp. 
Be aware that while powered you can change the settings in the menus but they will disappear as you turn of the device. 
Also, the last selected tool gets lost after a power cycle, which is usually stored on the SD-Card too.

As for the programming: You have to do it via the ST-Link (or "Upload" in VSCode / Platform.io), otherwise the starting address will be set wrong and the main routine may not get called. This will also overwrite the bootloader.
Also, with the SKR you're not able to use I2C for communication because the SKR has to be configured as master for the OLED.  For I2C communication you'll need to set it up as slave/servant, which would render the OLED unusable. You must user the serial connection.

davidcgu, 2020-01-03:
Great, for testing purposes will be fine for me until getting a new board, I should start writting at the standard address for the bootloader which I know it is 0x08000000.

I don't know why but have not being able to get st-link to work with VS while it works fine with St.link soft.

Regards

technik-gegg, 2020-01-04:
I needed to flash my ST-Link with the latest firmware before they got recognized by VSCode / openOCD.

davidcgu, 2020-01-04:
Thanks, my oled it's working know, so at least I know the board I made it's fine and will be ok to make some movement tests.

I already upgraded the firm of the ST-link to latest and still not recognized by VS.

I falshed with STM32 cube tool from ST-link starting on address 0x08000000 which is the one assigned originally for the bootloader and it's fine, the good point is that now I learn how to flash stock bootloader and marlin with the st-link so can recover the board to another use if required.

I have no idea how the system works and have a small question for you.... I have a BMG on direct drive and don't like the idea to move to bowden, my question is.... how exactly system works? there is a way that SMUFF does the load/unload and afterwards is disabled and works the direct drive extruder? should work both in paralel (smuff extruder and the one on direct drive)?

Thanks a lot for your time, much appreciated.

technik-gegg, 2020-01-04:
Great to hear it's working.

Yes, that's more or less the standard function. You connect all 3 steppers (or 2 and the servo) to it, configure such parameters as bowden tube length and the SMuFF will do exactly this on a Tx command.
It'll retract the current filament, change the Selector position, load the new filament and then release the filament (i.e. move the Revolver to home or open the servo). From there, your direct drive extruder has to take over and feed to nozzle.
That's the way the Prusa MMU does it and it also has some remarkable changes in their printers firmware as well to accomplish this task. For example: They order the MMU to feed filament until the filament sensor of the extruder triggers, then the printer send an **Abort** command. This makes it easier to handle when the bowden tube length changes but you need far more integration work on the printers firmware side. 

As I've already stated a couple of times: The major culprit here is that you have to feed the filament through one more set of gears. As you know, BMG has very tight tolerances and the only way to feed some bulged filament into it is: Not to have bulged filament! Means, you have to experiment a bit with filament tip shaping techniques.

That's one of the main reasons I'm recommending the bowden system. 

The other reason is: If you don't have a BMG direct drive system (i.e. something  MK8-ish), it doesn't make much sense going from the SMuFF BMG gears to something weaker and less reliable. 
But that's just my recommandation and nothing that's carved in stone.

closed 2020-01-06 by technik-gegg