technik-gegg/SMuFF-1.1 #7: [idea] klipper

Dario-Ciceri, 2020-04-26:
I would like to use this sistem on a voron 2.4 running klipper with a duet wifi. I will connect the motors and the endstop for this machine to my duet expansion board and I'm wondering if is it possible to use klipper to control this... I can try this out now because I don't own this machine and I don't have a Voron but after that I'll try

technik-gegg, 2020-04-26:
Well, I'm not sure why one would use Klipper on a Duet WiFi.
As far as I understand Klipper, it was meant to overcome the restrictions you'll get on an 8-Bit controller board, which is not being able to handle GCode at higher print speeds.
Klipper is based on the Raspberry, which is 32-Bit and thus has much more computing power than the average 8-Bit printer controller.

The Duet boards already are equipped with 32-Bit controllers and they do have the power to print with higher speeds right away.

That being said, as you plan to control the SMuFF hardware all through Klipper and the Duex, it sure is possible. You only have to set up your Klipper scripts so that they go through all the necessary steps to accomplish a tool change (i.e unload filament, move selector, load new filament).

Dario-Ciceri, 2020-04-26:
raspberry is 64bit too but that doesn't matter cause yes it's true I don't need to use klipper and multicolor/filament is a better choice at the moment

technik-gegg, 2020-04-26:
Just one thing you may have to take into consideration as we're talking of speed:
On my machines I'm able to print reliable with speeds up to 180mm/s. Everything that goes beyond will cause under extrusion because the extruder is not able to cope with higher speeds than that and I'm talking about Bondtech extruders.
So, even if you're able to speed computing up with 64-Bit, you'll still face the extrusion problem unless you use a completely different extrusion system.

So, If you want to drive the SMuFF without its controller but only via the Duet/Duex w/o Klipper, there're some guys who did that already. Head over to [this remix](https://www.thingiverse.com/thing:4208998) and see if that's of any use for you.
You might need to ask Codycato for his (Duet) scripts directly, since I'm not sure he published them.

Dario-Ciceri, 2020-04-26:
Don't worry I changed idea cause I figured out that I can't use smuff with directdrive and I want directdrive so I'll put smuff on my second printer which I use for mod testing 😄 
Thank you I'll let you know when I'll try on my hevo!!!

closed 2020-04-26 by Dario-Ciceri