technik-gegg/SMuFF-1.1 #8: Compatible LCDs

sl1pkn07, 2020-05-01:
Hi

im looking and buy the pieces for mount this awesome project!

but i have a lite questions

what type of LCDs is compatible with Smuff?

can work with SPI displays like the CR10/ENDER3 display? (10wires, display+encoder) (i have one wothout use)

if only use I2C, this display can work? https://es.aliexpress.com/item/32988162684.html

0.9" is little small for me (and +2" is quite expensive)

my project is based on ST NUCLEO-F103RB+CNC shield

greetings

technik-gegg, 2020-05-02:
Since you planning not to use an aftermarket controller (i.e. SKR mini) but the Nucleo development board, you can use pretty much any display you pick. 
Though, you have to check beforehand whether or not the display controller you pick is supported by the [U8G2 library](https://github.com/olikraus/u8g2/wiki) from Oliver Kraus, since this is the one the SMuFF firmware builds upon.
If it does, you only have to initialize the correct display (controller type) in SMuFF.cpp.
I'd say, chances are pretty good you'll be able to reuse your CR10 display (aka standard RepRap 12864 display).

I've used the I2C OLED for two reasons:
1. The SKR mini on board display connectors are driven through the SPI3 interface. U8G2 doesn't support this interface directly (only SPI1 and SPI2 in some cases).
This would have required to clone the library and patch it where needed, which is somehow neglecting the meaning of a "standard library".
2. The I2C OLED is pretty easy to integrate (4 pins only). Sure, it's physical size (0.9" or 1.3") isn't huge but it comes with a decent resolution of 128 x 64 px. 
In case of the SMuFF controller the physical size doesn't matter much, since you'll not be doing something major with it most of the time. It's being used only for setup and some status information display.

sl1pkn07, 2020-05-02:
Hi

the TFT i posted use the SSD1327 controller, and seems is supported by U8G2

i use the nucleo because have the same family of MCU (STM32F103XXX6) and i have already one

thanks

PD: a litle offtopic. you know this board? 

https://wiki.fysetc.com/AIO_II/
https://github.com/FYSETC/AIO_II

[image]

i think is perfect for you project. have 4 drivers (exist a module for 4xTMC2208 in UART), have integrated LCD, encoder, SD and reset button. also have a free Serial pin header (include I2C pinout)

technik-gegg, 2020-05-02:
Yes, the display is supported although, it's got twice the resolution in height (128px), so you might need to modify some of the screens of the SMuFF firmware.

The FYSETC controller is nice! Great find. 
It's a combination of the Wanhao i3 mini (first controller of the SMuFF) and the SKR - I might want one ;o)

sl1pkn07, 2020-05-02:
i have one, but i want to use for other project (DVD drive printer). sadly the missing servo. i'm not sure if can be used the fancy RGB port)

greetings

technik-gegg, 2020-05-02:
You can use the RGB if you get access to the signal before the MOSFETS.
That's the price you pay for highly integrated stuff...

sl1pkn07, 2020-05-10:
Hi, sorry for hihacked my own thread, bit i need little help with support for Nucleo-f103rb

<details>

~~~patch
diff --git a/Configs/NUCLEO_F103RB/SMUFF_Duet.CFG b/Configs/NUCLEO_F103RB/SMUFF_Duet.CFG
new file mode 100644
index 0000000..39770dc
--- /dev/null
+++ b/Configs/NUCLEO_F103RB/SMUFF_Duet.CFG
@@ -0,0 +1,70 @@
+{
+   "Serial1Baudrate":  56700,
+   "Serial2Baudrate":  56700,
+   "SerialDueBaudrate": 56700,
+   "ToolCount":                   5,
+   "BowdenLength":             620.0,
+   "SelectorDist":      34.0,
+   "LCDContrast":              190,
+   "I2CAddress":                  136,
+   "MenuAutoClose":       20,
+   "FanSpeed":                         50,
+   "DelayBetweenPulses": false,
+   "PowerSaveTimeout":         300,
+   "Duet3DDirect":      true,
+   "EmulatePrusa":      false,
+   "UnloadCommand":     "",
+
+   "Selector": {
+      "Offset": 0.5,
+      "Spacing": 21.0,
+      "StepsPerMillimeter": 80,
+          "Acceleration": 12000,
+          "MaxSpeed":  2000,
+          "MaxSpeedHS":  2500,
+          "InvertDir": false,
+          "EndstopTrigger": 1,
+      "StepDelay": 10
+   },
+   "Revolver": {
+          "StepsPerRevolution": 9600,
+      "Offset": 1275,
+          "Acceleration": 22000,
+          "MaxSpeed":  10000,
+          "ResetBeforeFeed": true,
+          "HomeAfterFeed": false,
+          "InvertDir": false,
+          "EndstopTrigger": 1,
+      "StepDelay": 5,
+      "Wiggle": false,
+      "UseServo": false,
+      "ServoOffPos": 30,
+      "ServoOnPos": 90
+   },
+   "Feeder": {
+          "ExternalControl": true,
+      "StepsPerMillimeter": 410,
+          "Acceleration": 20000,
+          "MaxSpeed":  1400,
+          "InsertSpeed": 20000,
+          "ReinforceLength": 0,
+      "UnloadRetract": 0,
+          "UnloadPushback": 0,
+          "PushbackDelay": 2.0,
+          "InvertDir": true,
+          "EndstopTrigger": 1,
+      "StepDelay": 3,
+      "EnableChunks": false,
+      "FeedChunks": 20,
+          "InsertLength": 5,
[... 404 lines cut ...]
+#ifdef __BRD_NUCLEO_F103RB
+U8G2_SSD1327_WS_128X128_F_HW_I2C display(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
+#endif
 
 ZStepper                steppers[NUM_STEPPERS];
 ZTimer                  stepperTimer;
~~~

</details>

~~~
Linking .pio/build/NUCLEO_F103RB/firmware.elf
/home/sl1pkn07/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/NUCLEO_F103RB/firmware.elf section `.rodata' will not fit in region `rom'
/home/sl1pkn07/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: region `rom' overflowed by 320 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/NUCLEO_F103RB/firmware.elf] Error 1
~~~

:cry:

how i can redute that 320Bytes?  in one of your post in thinkiverse (i don't rebember) you say i can put the json files into the headers, and i can get rid the SD code and json parser

how?

greetings and sorry

technik-gegg, 2020-05-10:
I don't think you'll get it 320 bytes down removing the JSON parser.
I'd rather say, shorten the texts in String.h, since those will be put into the ROM section as well.

To get rid of the JSON parser, you have to put your values as defaults into the smuffConfig structure in SMuFF.h and comment out the code in Config.cpp.

sl1pkn07, 2020-05-10:
with

~~~patch
diff --git a/include/SMuFFBitmaps.h b/include/SMuFFBitmaps.h
index a2ac3cd..42028e6 100644
--- a/include/SMuFFBitmaps.h
+++ b/include/SMuFFBitmaps.h
@@ -22,7 +22,11 @@
 
 #define logo_width  128
 #define logo_height 64
-static const unsigned char logo_bits[] PROGMEM = {
+#ifdef __BRD_NUCLEO_F103RB
+  static const unsigned char logo_bits[] PROGMEM = {
+  0x00, };
+#else
+  static const unsigned char logo_bits[] PROGMEM = {
   0x00, 0x00, 0xE0, 0xFF, 0x07, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xFC, 0xFF, 0x1F, 0x00, 0x00, 0x00, 
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x80, 0x7F, 0x00, 
@@ -109,5 +113,5 @@ static const unsigned char logo_bits[] PROGMEM = {
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
   0x00, 0x00, 0x00, 0x00, };
-
+  #endif
 #endif
diff --git a/src/SMuFF.cpp b/src/SMuFF.cpp
@@ -109,5 +113,5 @@ static const unsigned char logo_bits[] PROGMEM = {
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
   0x00, 0x00, 0x00, 0x00, };
-
+  #endif
 #endif
~~~

~~~
Checking size .pio/build/NUCLEO_F103RB/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  53.9% (used 11032 bytes from 20480 bytes)
Flash: [==========]  99.5% (used 130368 bytes from 131072 bytes)
================================================================================================================== [SUCCESS] Took 4.79 seconds ==================================================================================================================

Environment       Status    Duration
----------------  --------  ------------
WANHAO_I3_MINI    IGNORED
BIGTREE_SKR_MINI  IGNORED
NUCLEO_F103RB     SUCCESS   00:00:04.785
================================================================================================================== 1 succeeded in 00:00:04.785 ==================================================================================================================
~~~

should be ok?

technik-gegg, 2020-05-10:
WTF?

technik-gegg, 2020-05-10:
Well, ok ;o)

sl1pkn07, 2020-05-10:
the code is not for merge to upstream, so don't worry xd

sl1pkn07, 2020-05-13:
well, the branch experimental overflow the nucleo by 1264 bytes

well, time to mod the nucleo! (STM32F103RET6 incoming!)

closed 2020-05-17 by technik-gegg