technik-gegg/SMuFF-1.1 #15: Cannot compile Fysetc AIO II version

oengst, 2020-10-18:
Hello, 

today I tried to compile the firmware for the Fysetc AIO II board.

When I try to compile it I get some errors about some settings not being defined.

`src\GCodes.cpp: In function 'bool M569(const char*, String, int)':
src\GCodes.cpp:1541:26: error: 'STALL_X_PIN' was not declared in this scope
         if(param == 0 && STALL_X_PIN != -1)

                          ^~~~~~~~~~~
src\GCodes.cpp:1546:12: error: 'STALL_X_PIN' was not declared in this scope
         if(STALL_X_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp:1558:26: error: 'STALL_Y_PIN' was not declared in this scope
         if(param == 0 && STALL_Y_PIN != -1)

                          ^~~~~~~~~~~
src\GCodes.cpp:1563:12: error: 'STALL_Y_PIN' was not declared in this scope
         if(STALL_Y_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp:1575:26: error: 'STALL_Z_PIN' was not declared in this scope
         if(param == 0 && STALL_Z_PIN != -1)

                          ^~~~~~~~~~~
src\GCodes.cpp:1580:12: error: 'STALL_Z_PIN' was not declared in this scope
         if(STALL_Z_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp: In function 'bool M914(const char*, String, int)':

src\GCodes.cpp:1728:12: error: 'STALL_X_PIN' was not declared in this scope

         if(STALL_X_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp:1733:12: error: 'STALL_X_PIN' was not declared in this scope
         if(STALL_X_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp:1749:12: error: 'STALL_Y_PIN' was not declared in this scope
         if(STALL_Y_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp:1754:12: error: 'STALL_Y_PIN' was not declared in this scope
         if(STALL_Y_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp:1770:12: error: 'STALL_Z_PIN' was not declared in this scope
         if(STALL_Z_PIN != -1)

            ^~~~~~~~~~~
src\GCodes.cpp:1775:12: error: 'STALL_Z_PIN' was not declared in this scope
         if(STALL_Z_PIN != -1)

            ^~~~~~~~~~~
*** [.pio\build\FYSETC_AIOII_SD\src\GCodes.cpp.o] Error 1`

Am I missing a setting or is that an issue with the project?

Thanks a lot.

technik-gegg, 2020-10-19:
You're not missing anything, that's a bug!

Please add these lines to  the **Pins.h** for the FYSETC_AIOII:
```
#define USB_CONNECT_PIN     -1
#define SD_DETECT_PIN       -1

#define STALL_X_PIN         -1
#define STALL_Y_PIN         -1
#define STALL_Z_PIN         -1
```
Also, in order to compile successfully, please modify your **platformio.ini** and add the "# add this" lines:

```
[env:FYSETC_AIOII_SD]
platform         = ststm32
framework        = arduino
board            = genericSTM32F103RC
board_build.core = maple
lib_deps         = ${common.lib_deps}
                  # add this
                   ${TMC.lib_deps}

build_flags      = ${common.build_flags}
                   ${USB.build_flags}
                 # add this
                   ${TMC.build_flags}

                   -I include/FYSETC_AIOII
                   -D __BRD_FYSETC_AIOII
                   -D __STM32F1__
                   -D USE_RGB_BACKLIGHT
build_unflags    = ${common.build_unflags}
extra_scripts    = include/FYSETC_AIOII/FYSETC_AIOII_SD.py
```

I will correct this for 2.12.

oengst, 2020-10-19:
Hi, 

thanks for the information. 
With that data I am able to compíle it. I still have an issue uploading it to the device but that might be related to my PlatformIO setup. 

The upload via SD card is not doing anything. 
So I downloaded the STM32Flash tool . 
If I use that one it stops with the information: "Program Space Must In either (0x08000000-0x0807FFFF) or (0x20000000-0x2000FFFF)"

`DTR be Low(-3--12V),Reset
RTS be High(+3-+12V),enter ISP
...Delay 100ms
DTR be High(+3-+12V),Reset Released
RTS keep High
Connectting...4, Received:79 1F 
Connect Ok @COM3@460800bps,@641ms
BootLoader Version:2.2
PID:00000414  STM32F10xxx_High-density
OptionBytes readout:
A55AFF00FF00FF00FF00FF00FF00FF00
96bit unique device ID:
[38FFDC05504D333238850243]
[05DCFF38 32334D50 43028538]
FLASH ROM size:256KB
SRAM size:65535KB(For reference only!!!)
Program Space Must In either (0x08000000-0x0807FFFF) or (0x20000000-0x2000FFFF)
`

That looks like the python scripts in the include folder are not running. So I tried to run external (as a newbie to python) but it fails on the 

`Import("env")`

line. Research in internet did not reveal to me what I need to use to make imports work that way. 

Maybe you can help me with a hint to get around this?

Thanks  a lot.

technik-gegg, 2020-10-19:
Be careful with the Flash tool. If not configured properly, it might erase your bootloader!

All environments named **_SD** are supposed to be flashed using the SD-Card.
Copy the **firmware.bin** file from the **SMuFF\.pio\build\FYSETC_AIOII_SD** folder onto the SD-Card, insert it into the AIOII, reset the device and wait a couple of seconds.
Please notice that, after the firmware is being flashed, the SMuFF will start up and try to read the SMUFF.CFG file from the SD-Card. So you have to copy that one as well.

Just for reference: [Compile the firmware](https://sites.google.com/view/the-smuff/how-to/tutorials/compile-the-firmware
)

oengst, 2020-10-20:
Thanks for your support. 

I was now able to compile the firmware from the experimental branch and upload it to the board. I just needed to flash the original bootloader from board manufacturer since there was already a Marlin on the device when I bought it.
And that did obviously not upload a new firmware.bin when available at startup.

Is it mandatory to have everything wired up to be able to do anything in the smuff controller? 

My actualy situation is like in this private shared video (only available with link)

https://www.youtube.com/watch?v=wIhndu7ZMWw

After starting it does some beeps, reads the configuration and then opens settings. After that it stops doing anything. No reaction on controller turn or click.

Would it be easier if I use a kind of "standard set" to wire the Smuff up? Which board would that be? Would that make things easier for me?

Thanks a lot.

technik-gegg, 2020-10-21:
Yeah, that seems to be an odd issue on the AIOII.  
Maker @sl1pkn07 found out, that the encoder needed additional [pull-up resistors on the encoder pins](https://github.com/technik-gegg/SMuFF-1.1/pull/11#issuecomment-649840817) to make that thing work.

Well, for the start all you need is the board, display, SD-Card and firmware but you won't be able to do much, except changing the settings. If you want to play around, I'd recommend connecting at least two stepper motors (X and Z), two endstops (for X and Z) and one servo. 
Also, I'd recommend using a USB to TTL serial interface and connecting the it to the SMuFF, so you can control it via a terminal app by [sending GCodes](https://sites.google.com/view/the-smuff/tips-hints/talk-to-the-smuff?authuser=0).

> Would it be easier if I use a kind of "standard set" to wire the Smuff up? Which board would that be? Would that make things easier for me?

Not sure what you mean by "standard set"...

sl1pkn07, 2020-10-25:
tested in my unit (with pullups in the enc pins). same problem like @oengst  (freeze in the settting menu)

about this board, by default comes without bootloader, you must burn it (the bootloader) before use the flash by SD feature

greetings

oengst, 2020-10-26:
> Not sure what you mean by "standard set"...

I wonder which is the newest board / screen combination that is supported and known as working perfectly. 

Thanks :-)

technik-gegg, 2020-10-27:
Oh... that.
Well, there's not much of a choice here. The initial (2.x) board, the SKR mini V1.1 isn't available anymore (or at least hard to get). So, you'll probably go for the SKR E3 DIP (V1.1).
Basically, all SKR E3 versions are supported (E3 - **not DIP** - V1.2 and V2.0).  Though, with the V1.2 and 2.0 you'll get all the stepper drivers soldered directly onto the board, so there's not much room to tinker with.

As for the display: I myself still prefer the [DIY OLED](https://sites.google.com/view/the-smuff/how-to/tutorials/build-the-custom-display?authuser=0) solution over all others. 
In the next release of the firmware, [this OLED module](https://sites.google.com/view/the-smuff/work-in-progress?authuser=0#h.1efg2hp8wplx) will be supported as well. Beside a few buttons more, this comes with the advantage of a 1.3" display which is easier to read and you won't have to DIY.

closed 2020-11-03 by technik-gegg

sl1pkn07, 2020-11-16:
as note. the version 2.09 is the latest version that works with the AIO unit. but need a little tweak not added in the code in that version

~~~
diff --git a/src/SMuFF.cpp b/src/SMuFF.cpp
index 37cfee7..9d6b0b1 100644
--- a/src/SMuFF.cpp
+++ b/src/SMuFF.cpp
@@ -306,6 +306,12 @@ void initFastLED() {
 void setup() {
 
 #ifdef __STM32F1__
+
+  // Disable JTAG pins for use in general purposes
+  #ifdef __BRD_FYSETC_AIOII
+    disableDebugPorts();
+  #endif
+
   #ifndef USE_TWI_DISPLAY
     #ifdef STM32_REMAP_SPI
     afio_remap(AFIO_REMAP_SPI1);  // remap SPI3 to SPI1 if a "normal" display is being used

~~~

also, the LCD color sequence not work al all:

Menu            Real Color 
Cyan       ->  Blue
Magenta ->  Magenta
Yellow     ->  Cyan
White     ->  White
Black      ->  Black (no backligth)
Red        ->  Red
Green    ->  Green
Blue       ->  Yellow  

greetings

technik-gegg, 2020-11-17:
Thank you Gustavo.
I'll add the SPI remapping back in 2.14 and correct the color settings.