technik-gegg/SMuFF-1.1 #18: TWI problem

luismello, 2020-11-08:
After turn ON, it has continuous beep, I checked and the file size is 2200, but the SMUFF.CFG file has 2378, I changed the file size limit and Smuff turn ON.

But the Main Menu doesn't appear.

technik-gegg, 2020-11-09:
Have you read the [README of this release](https://github.com/technik-gegg/SMuFF-1.1/tree/SMuFF-2.0-(Experimental)#recent-changes)? There have been a lot of changes for menus and config files.

luismello, 2020-11-09:
hi

I put Menus, Options and Sound, but the menus still have a problem

[image]
[image]

technik-gegg, 2020-11-10:
I'm working on it, please stay tuned.

Nice wiring btw. so... colorful ;o)

luismello, 2020-11-10:
Hi TG

I will buy the BTT Mini12864 V1.0.

I did with the threads I had at home, hehe

technik-gegg, 2020-11-11:
@luismello 
Checked in 2.13. Please try it and let me know if there are still issues.

luismello, 2020-11-14:
Hi TG

Version 2.13 did not solve my problem with Oled.

technik-gegg, 2020-11-15:
Is your OLED basically working and has a distorted screen or does it freeze up at some point?

closed 2020-12-22 by technik-gegg