technik-gegg/SMuFF-1.1 #19: Not working with SKR mini E3 V1.2

nasrabq, 2020-12-01:
It build successfully but when flashing it to SKR mini E3 V1.2 using the SD card, the LCD shows only backlight.
I am using the Creality LCD, I have verify that both the mainboard and the LCD are working as I upload Marlin software and they are working OK.
I have changed the display in Platformio.ini and then build the BIGTREE_SKR_MINI_E3_SD, is there anything I missed?

[image]

technik-gegg, 2020-12-01:
Which firmware version have you been using?

nasrabq, 2020-12-01:
I am using SMuFF-1.1-SMuFF-2.0-Experimental

technik-gegg, 2020-12-01:
Yeah, that's for sure. 
What does the Config.h say?
Also, do you hear a tune (melody) after startup?
And, last but not least, have you copied all the files and folders to the SD-Card as well?

nasrabq, 2020-12-01:
What does the Config.h say? **I have not change anything in this, it is the same as downloaded.**
Also, do you hear a tune (melody) after startup? **No sound at all**
And, last but not least, have you copied all the files and folders to the SD-Card as well? **I copied the configuration files from \Configs\SKR_mini folder only, do I need to copy other files**

technik-gegg, 2020-12-01:
In the Config.h is the latest VERSION string.
Yes, you do, [have a look](https://discordapp.com/channels/741179485474521110/742611826264506368/780830875582464070) (which is btw. also described in the README.md).

nasrabq, 2020-12-01:
VERSION_STRING    "V2.13"
I have copied the below folders in the SD card:
- help
- menus
- options
- sounds
- test

also copied the below files:

- firmware.bin
- MATERIALS.CFG
- SERVOMAP.CFG
- SMUFF.CFG
- TMCDRVR.CFG

but still did not get anything in the LCD also no sound

technik-gegg, 2020-12-01:
ok, the reason is: You need VERSION 2.14 (which is the latest published).

Be aware that you might not be able to select any menu item because the encoder button seems not to be working on some Creality displays (without modification). It does work on mine (original Ender-3 Pro display) but I've been told there are some out there who don't.

nasrabq, 2020-12-01:
Thank you, It is working now.

technik-gegg, 2020-12-01:
The encoder button as well?

nasrabq, 2020-12-01:
Yes, the encoder button is working

technik-gegg, 2020-12-01:
Good to know. Thanks!

nasrabq, 2020-12-01:
I have another issue with the display, It looks unreadable:
[image]

technik-gegg, 2020-12-01:
You're the second one having this issue...

I'm not quite sure, but I'm pretty convinced that this issue relates to your SD-Card. Menus and options will be read from SD-Card (hence you had to copy the folders). I bet the start screen on your display looks pretty normal!?

Can you format your SD-Card and give it another try (or better yet, use a fresh one).

nasrabq, 2020-12-02:
I have tested the with multiple SD-Cards but all not displaying anything and multiple sizes (128MB, 1GB and 8GB)
I have flashed Marlin and the SD-Card working OK, I can see the STL files in the SD-Card.

technik-gegg, 2020-12-02:
Ok, I'm stumped. I can't reproduce that with my  equipment. What about the main screen and the tools menu, are they showing correctly?

nasrabq, 2020-12-02:
I reflash it and now shows nothing, the firmware isflashed but because the menus is in the SD it is not displayed

technik-gegg, 2020-12-02:
Make sure you don't have copied the config files from 2.13, they don't work anymore.
At this stage it's hard to tell what went wrong. The only way to check is to connect the controller via TTL to USB on TFT connector to the PC and watch the debug outputs in a terminal app ([here's the how to](https://sites.google.com/view/the-smuff/tips-hints/talk-to-the-smuff?authuser=0)).

technik-gegg, 2020-12-06:
Good news! Found a bug in my .editorconfig file. 

The .editorconfig of the SMuFF-Project needs to be set to:
```
[*.mnu, *.opt]
charset = utf-8
end_of_line = crlf
```

Please change it manually or get the latest version from my Github repository and make sure that your **.mnu** and **.opt** files on the SD-Card are ending with CR + LF (_0D 0A_) instead of LF only (_0A_). You may have to use a HEX-Editor to check the files.
Otherwise the loadMenu function will not parse the files correctly and thus menus will look pretty much distorted.

closed 2020-12-06 by technik-gegg

technik-gegg, 2020-12-08:
I have to correct myself. 
Obviously this was not the cause for this kind of error. There's now a new FW version (2.16) released, which compiles against the STM32 V10.0.1 framework. This framework has got some fundamental changes in the SD-Card Library (most probably introduced already in V9.0.0), which might explain this strange behavior.
Please use the latest version and tell me whether or not this neutralizes the "faulty display" error.